NilsRo/OctoPrint-SlicerEstimator #1: Please add support for Simplify3D

leeph, 2021-02-25:
As per title, it is still a very popular slicer.

NilsRo, 2021-02-25:
Hi. Could you send me the format of the Gcode reflecting the remaining time. Every sliver seems to have its own format which have to be handled by a RegEx. In common it is shown in M117 or M73 commands.

leeph, 2021-02-25:
Thank you for looking into this.

Here is a real-life Simplify3D GCode output. I did a quick search but couldn't find the two commands you mentioned, so I have no idea how it does the time estimation; I hope it's still possible. It does have time estimation comments at the end.

[Curtain Bracket Body.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/6042860/Curtain.Bracket.Body.zip)

NilsRo, 2021-02-25:
I did also not find any information about the remaining time during print. You should look in Simplify3D if a feature could be activated. In Cura and as I know also in Slic3er PE it is not active by default. Look for something that will print it to the printer screen.

NilsRo, 2021-02-25:
> Here is a real-life Simplify3D GCode output. I did a quick search but couldn't find the two commands you mentioned, so I have no idea how it does the time estimation; I hope it's still possible. It does have time estimation comments at the end.

I did some google research and it seems to be a plugin available. But the post is older so perhaps they added this feature. 
https://forum.simplify3d.com/viewtopic.php?t=9019

NilsRo, 2021-02-26:
I spend some thoughts about it it should be possible to support Simplify3D. But I have to rewrite a new parsing as the actual routine is build to search the remaining time in the commands send to the printer. With Simplify3D I have to search the whole file for the estimation and then using percentage done to calculate the remaining time. Should be somehow possible...we will see.

NilsRo, 2021-02-27:
@leeph : I updated most of the plugin to parse a file for a string and do estimation with help of print progress. In my test instance it is working fine with your file. Also I need from you a long print runtime like slicing a cube scaled up to a large size with infill 100 (you did not need to print it). I know it from some other tools that the string is  perhaps changing to 2 days 2 hours 10 minutes. That you must test as I did not have installed Simplify3D. 

;   Build time: 1 hour 12 minutes

NilsRo, 2021-03-02:
Simplfy3D is not available in Release 0.5

closed 2021-03-02 by NilsRo