NilsRo/OctoPrint-SlicerEstimator #3: support for octoprint api

adcurtin, 2021-03-01:
the octoprint api reports the print time left. It would be awesome if this plugin could overwrite the value reported by the api (maybe even change the printTimeLeftOrigin to SlicerEstimator or similar).
  
  ```
"progress": {
    "printTimeLeft": 0,
    "printTimeLeftOrigin": "estimate"
}
```

NilsRo, 2021-03-01:
Hi,

actually I also use "estimate" as the Origin and as I understand the API behaviour the print time read out of the slicer is still available in the API but it is not really visible the the receiver of the Rest Call. To have a clear view on this I changed the Origin to "slicer" in the next version. I will left this incident open until it is released.

NilsRo, 2021-03-01:
@adcurtin: If you like you can install branch https://github.com/NilsRo/OctoPrint-SlicerEstimator/tree/Simplify3D%2C-Cura-Native-Support 

I am waiting for some information for Simplify3D and will release it in the near future.

closed 2021-03-01 by NilsRo

adcurtin, 2021-03-01:
ah fantastic. I'll give that a shot. What s3d info do you need?

Also, it's prusaslicer now, not slic3r prusa edition. the slic3r pe setting didn't seem to work for me either (for prusaslicer), so I'm still playing around and getting everything to work.

NilsRo, 2021-03-01:
I did not installed PrusaSlicer (from Github Repo it seems still Slic3r) and got the format from a demo gcode file and a description. So perhaps there is a mistake from my side. Looking forward to your testing. If you like simply attach some gcode files than I can check this in the coding.

For S3D I got some examples from somebody in Octoprint forum so I can check my implementation finally. If I can fix PrusaSlicer soon it would be nice for the next release.