NilsRo/OctoPrint-SlicerEstimator #5: No configuration GUI shown...

energiebrand, 2021-03-02:
I have installed the PlugIn but there is no configuration GUI shown.
See the attached Releases and picture of the GUI. 

I did remove the installation and then re-installed but with the same result.

[image]

[image]

NilsRo, 2021-03-02:
This was a bug 4 hours online for new installations. The defaults settings was wrong (and during my tests I did not do a clean install to my regret). It is fixed and released in 0.51 since 37 minutes.

energiebrand, 2021-03-02:
Ok, thanks a lot for fast feedback! Will test a new installation tomorrow, just started printing :-)

energiebrand, 2021-03-03:
Ok, working now! Thanks a lot!

closed 2021-03-03 by NilsRo