NilsRo/OctoPrint-SlicerEstimator #8: prusaslicer files give estimate intermittently, then reverts to null

adcurtin, 2021-03-05:
I've got the plugin set to prusaslicer, and my estimate is alternating between the correct time remaining, and null. 

it looks like about every minute, the estimate is correct for 10 seconds, then null for the rest.

NilsRo, 2021-03-05:
Hi. 

Can you upload the gcode file or a octoprint log with the plug-in debug mode set. I sliced a testfile with the newest version of PrusaSlicer with run without issues but will check again. 

Nils

adcurtin, 2021-03-05:
how do I set debug mode on the plugin?

NilsRo, 2021-03-05:
It is a general Octoprint function. You have to enable debug for octoprint.plugin.slicerestimator or something like this. After you reproduced the issue download octoprint.log from the same Place.

https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

Activate ˋvirtual printerˋ if you like to do a simulated print. 

As everything is contained in the Gcode file it would be another option.

cyrusjk, 2021-03-06:
I am using PrusaSlicer and I am also getting nulls displayed
[image]

NilsRo, 2021-03-06:
@cyrusjk  Interesting, I cannot reproduce it with PrusaSlicer. Could you upload the debug logs and the GCODE file? Description you can find in the post before.

Do you perhaps use another plugin doing something special to the remaining printing time send via M73?

cyrusjk, 2021-03-06:
I will try with a smaller file once my current print is done. In the meantime, here is a shot of my current listed plugins:

[image]

NilsRo, 2021-03-06:
@cyrusjk I had installed also most if them. 2 Plugins injecting GCODE commands regarding print time to the printer commands. Detailed Progress and Octoprint-Display-ETA. You should disable the injection or delete one of them as both injecting the same which could be a problem in general. I tested Octoprint-Display-ETA which works fine.

By the way, PrusaSlicer would be the 3rd tool injecting progess information. As on the "Detailed Progess" Homepage a rotation is mentioned perhaps this is the cause for the intermediate display described by @adcurtin    If this is confirmed I can change two things...try to get the GCODE before it is changed by the plugins or harden the RegEx to get the information from the GCODE.

cyrusjk, 2021-03-06:
Removed "Detailed Progress" and added in "PrintTimeGenius Plugin", and now I am seeing estimates intermittently. They are rounded off to the minute.

with:
[image]

without:
[image]

NilsRo, 2021-03-07:
Without the debug log I cannot do anything. I need it for the analysis of the issue. But what you described shows we are on the right way.

cyrusjk, 2021-03-07:
Just did a calibration cube.

[debug_calibrations.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/6097204/debug_calibrations.zip)

NilsRo, 2021-03-07:
Hi @cyrusjk 

I found the issue for this. There are unknown M73 commands in addition to the commands generated by PrusaSlicer.
[image]

I still added to logging for this but the exception handling was wrong. So unknown commands run into the - estimation.

You can download a correction here and test it: https://github.com/NilsRo/OctoPrint-SlicerEstimator/archive/Development.zip
Uninstall the plugin before and then use the link for installation in the plugin manager. It should show version 0.60 then. It also includes an update of the file browser if you upload a file to OctoPrint in its first version.

If you like to do a "virtual" print you can simply activate the virtual printer like shown here and connect to it instead:
[image]

cyrusjk, 2021-03-07:
Success! That does seem to remedy the issue for me using the virtual printer. 

It would be nice to have seconds as well as minutes, but I know that the data PrusaSlicer provides doesn't have that.

NilsRo, 2021-03-07:
Perfect. 

The estimation handler is not called that often that it makes sense to do a calculation to fill the gap. But I will check this for a future enhancement, it should not be very complex to implement.

closed 2021-03-07 by NilsRo

adcurtin, 2021-03-07:
ah yeah the M73 Q and S is for stealth mode, and P and R are for regular mode.