NilsRo/OctoPrint-SlicerEstimator #11: version 0.53 > version 0.6

adcurtin, 2021-03-09:
I just manually uninstalled and reinstalled the extension to get version 0.6. Now, the plugin updater is telling me there's an update from version 0.6 to version 0.53

NilsRo, 2021-03-09:
That is sometimes the case of the local plugin repo cache is not updated. In the central plugin repo the new version is still recognized. https://plugins.octoprint.org/plugins/SlicerEstimator/

I will keep an eye on it but I think it will be gone on its own if the cache is refreshed.

adcurtin, 2021-03-09:
even force refreshing the update cache still seems to think 0.53 > 0.6. 

i'm wondering if some things (like plugin install, and the central plugin repo) use date, and show the most recent dated version as latest, but the updated actually does version checking and does it differently.

NilsRo, 2021-03-09:
Changed versioning scheme to 1.0.0 as it is better supported. Octoprint will order the versions which leads into a issue with 0.53 and shorter 0.6.

adcurtin, 2021-03-10:
updated and everything's working perfect, thanks again!