NilsRo/OctoPrint-SlicerEstimator #20: Just getting green dot, no checkmark.

chrishilbert, 2021-03-29:
Running PrusaSlicer 2.3.0 with a Prusa mini on OctoPrint 1.5.3.  Plugin updated today to current version.  Only seeing green dot, no checkmark.

[image]

NilsRo, 2021-03-29:
Hi Chris.

A green dot is a very good average estimation. If you enabled this checkbox it will be used instead if available. 
[image]

But perhaps there is something wrong with the exchange of the checkmark as there was a Syntax error I had to correct at exactly this point. Can you enable debug log and attach the octoprint.log file. I can check the behaviour of the plugin then. Also a list of installed plugins would be nice to check if there could be some conflicting. I printed 3 jobs with the actual version but sliced with Cura without issues. If you like to simulate a print simply enable "virtual printer", should be the last entry in the plugins and connect to port VIRTUAL. So you have not to waste material.

[image]

chrishilbert, 2021-03-29:
I do have that box checked.  Do you still need the logs if it's operating as expected then?

NilsRo, 2021-03-29:
No, everything is like expected then. 

You can find details on the Github Homepage. The average estimation is not a real estimation it is an average of the last real prints if you printed an object more than once. It does also include heatup time so it will be different from slicer estimation. The slicer does not know how long the printer will heat up...so you can set it like your need...with heatup and based on history or slicer estimation everytime.   If you preheat the printer "sometimes" the average estimation could be slightly off.

closed 2021-03-29 by NilsRo