NilsRo/OctoPrint-SlicerEstimator #22: Breaks other plugins when no M73 is present?

codingcatgirl, 2021-04-05:
Hey,

when uploading a gcode file without M73 tags, the `['analysis’]` item seems to be set to None, which confuses the spoolmanager plugin.

This doesn't seem to be expected behavior.

codingcatgirl, 2021-04-05:
Specifically, it becomes impossible to load the file. Even without SpoolManager, the following exception is raised:

```
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1539, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 802, in gcodeFileCommand
    printer.select_file(filenameToSelect, sd, printAfterLoading, user)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 680, in select_file
    tags=kwargs.get("tags", set()) | {"trigger:printer.select_file"},
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 1599, in selectFile
    filename, self._currentFile.getFilesize(), False, user=user
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1545, in on_comm_file_selected
    self._setJobData(full_path, size, sd, user=user)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1282, in _setJobData
    and "estimatedPrintTime" in fileData["analysis"]
TypeError: argument of type 'NoneType' is not iterable
```

NilsRo, 2021-04-06:
Hi,

it creates an OctoPrint standard estimation and only if a slicer estimation is found it will overwrite the time needed. All other information is used from OctoPrint. Also there is an if statement that avoids overwriting if estimation is none. This works fine here but perhaps I oversee something special. Could you upload a Gcode file showing this behaviour and your plugins you are using.

IvanVolosyuk, 2021-04-06:
I have a minimal gcode I was using for quick test of my printer. I don't see upload file button, but the gcode is pretty small, so I can paste it here:
```gcode
;Start

; Speed 50 mm/s
G0 F3000
; Speed 10 mm/s
G1 F600

; Fan 50%
;M106 127

M900 K0.16

; Absolute move away
G90
G0 Z20
G0 GX20 Y120 Z20 F3000

; Set extruder position
G92 E0
G1 E5 F600
G0 X70 Y120 Z0.3 F3000
G0 X110 Y120 Z0.3 E5.1

; Relative position to all
G91

G1 X50 Y0 E2.49 F600
G0 X0 Y2.0
G1 X-50 Y0 E2.49 F600
G0 X0 Y2.0
G1 X50 Y0 E2.49 F600
G0 X0 Y2.0
G1 X-50 Y0 E2.49 F600
G0 X0 Y2.0

; Absolute position to all
G90
; Lift up
G0 Z100

; Fan off
;M107
```

NilsRo, 2021-04-06:
I created an updated version you can install from here. Please check if the error is fixed there. If this is fine it will be the same version released so you have not to change anything after.

https://github.com/NilsRo/OctoPrint-SlicerEstimator/archive/refs/heads/Development.zip

BTW. You can uploads files by dragging them to the text-field also with a large size.

IvanVolosyuk, 2021-04-06:
I'm in the middle of a few hours long print. I can only test this in approx. 20 hours. I wonder if @codingcatgirl can test the fix before that.

codingcatgirl, 2021-04-06:
I'm in a long print right now as well, but i'll test it as soon as i can (and as soon as i remember).

IvanVolosyuk, 2021-04-07:
It seems like it works now for me. Thanks for the fix! I remember without the fix I was unable to start a print because of the exception. Now it works fine.

NilsRo, 2021-04-07:
Thanks for testing. Published the fix now with release 1.0.3.

closed 2021-04-07 by NilsRo