NilsRo/OctoPrint-SlicerEstimator #26: Filelist is broken with lastest version

codingcatgirl, 2022-01-04:
I see that the last two updates brought some great features and i'm excited to use them in some way soon!

However… I had to disable this plugin after the latest update because the file list is not showing my files any more (just one file gets half-rendered). Disabling the file list feature in the settings does not solve the problem.

Here's an error that gets thrown. 

> packed_libs.js?1fe9f29c:132 Uncaught RangeError: Unable to process binding "template: function(){return { name:$root.templateFor($data),data:$data} }"
> Message: Unable to process binding "css: function(){return { disabled:!$root.enableAdditionalData($data)} }"
> Message: Maximum call stack size exceeded
>     at Function.keys (<anonymous>)
>     at FilesViewModel.self.filesViewModel.slicerEnableAdditionalData (packed_plugins.js?1e391fc5:2843)
>     at FilesViewModel.self.filesViewModel.slicerEnableAdditionalData (packed_plugins.js?1e391fc5:2846)
>     at FilesViewModel.self.filesViewModel.slicerEnableAdditionalData (packed_plugins.js?1e391fc5:2846)

Here's the list of plugins i have installed and enabled.

> Action Command Notification Support (bundled)
> Action Command Prompt Support (bundled)
> Announcement Plugin (bundled)
> Anonymous Usage Tracking (bundled)
> Application Keys Plugin (bundled)
> Arc-Welder (1.0.0+u.bb71e8f)
> Backup & Restore (bundled)
> Bed Cooldown (0.4.1)
> Bed Visualizer (1.1.1)
> Cancel Objects (0.4.6)
> Core Wizard (bundled)
> Cost Estimation (3.4.0)
> Error Tracking (bundled)
> Event Manager (bundled)
> File Check (2021.2.23) (bundled)
> FileManager (0.1.6)
> Firmware Check (2021.10.11) (bundled)
> GCode Viewer (bundled)
> Logging (bundled)
> MQTT (0.8.12)
> Octoprint Display ETA (2.1.0)
> Pi Support Plugin (2021.10.28) (bundled)
> Plugin Manager (bundled)
> PortLister (0.1.10)
> PrintJobHistory (1.15.2)
> Simple Emergency Stop (1.0.5)
> Slicer Print Time Estimator (1.1.0)
> Slicer Thumbnails (1.0.0)
> SlicerSettingsParser (2.0.1)
> SlicerSettingsTab (1.0.3)
> Software Update (bundled)
> SpoolManager Plugin (1.5.0)
> UI Customizer (0.1.8.1)
> Z Probe Offset (0.5)

I have tried disabling both UI Customizer and Slicer Thumbnails becausae those seemed like the most obvious candidates to cause trouble with this new feature but the problem persists.

Happy to provide more information as needed

NilsRo, 2022-01-04:
@codingcatgirl  Can you try the development version available here: https://github.com/NilsRo/OctoPrint-SlicerEstimator/archive/refs/heads/Development.zip

Simply install it via URL over the actual plugin. I implement some changes that should increase compatibility and fixed a bug with recursion that perhaps should caused the issue you reported.

codingcatgirl, 2022-01-05:
With the development version, the issue doesn't happen – both with the setting enabled and disabled. Thanks! :)

NilsRo, 2022-01-05:
The update handling will be fine with that version, so there is no need to do anything now.

closed 2022-01-05 by NilsRo

oerkel47, 2022-01-05:
@NilsRo I had the same problem but coupled with some additional confusing thing.
In my case it installed 1.1.0 and 1.1.1rc right after that. I don't even know why it installed the rc version by itself, so 1.1.0 might have enabled dev channel by itself.  However, even with that 1.1.1RC version the problem was there (in fact I didn't get to use the 1.1.0 version). I uninstalled and reinstalled the 1.1.1 from the repo since and now it's fixed and plugin has no dev channel anymore. 

Since this could potentially have happened to others you might want to know.

NilsRo, 2022-01-05:
Interesting... In 1.1.0 I introduced a development channel and it should ignore RC releases. But I decided to go a different way with 1.1.1 but nowhere it should install the release candidates without allow it without changing to the new channel. Anyhow, perhaps this caused the problem...but it should be solved with the actual release I think.