NilsRo/OctoPrint-SlicerEstimator #27: Incorrect parsing of metadata

j0dan, 2022-01-19:
I have this in my gcode:
```
;Slicer info:filament_settings_id;Prusament ASA
;Slicer info:print_settings_id;0.2mm SPEED (dan)
```

But Slicer Estimator parses it as:
[image]
```
ament_settings_id
print_settings_id
```

NilsRo, 2022-01-19:
Please switch to development channel in the plugins settings, update from there and report back. I pushed a fix that should correct this.

j0dan, 2022-01-20:
Same problem after switching to channel and updating.

Thanks for looking at it.

NilsRo, 2022-01-20:
Did you uploaded the file again? Overwriting it is fine. The metadata is read only on upload where I found a bug in the string split.

j0dan, 2022-01-20:
Your release just now for 1.2.2 fixed it.

NilsRo, 2022-01-21:
Mh, that is strange as I corrected it only in the upload routine and the version change comes next for preparing the release...

But very fine if it works now. Could you anyhow upload the octoprint.log. Perhaps I can see something that is going not as expected is it should not heal by itself without uploading. :-)   And you can switch back to normal release channel. I published version 1.2.2 also there.

closed 2022-01-22 by NilsRo