NilsRo/OctoPrint-SlicerEstimator #28: Display time until next tool change / M600

codingcatgirl, 2022-04-26:
This feature seem slightly out of scope for this plugin, but let me explain why it might not be.

When using an MMU (tool changes using T0, T1…) or manual tool changes using M600 it is useful to know how much time is left until the next tool change because you might need to be present for it.

There is currently an octoprint plugin for this, but it requires Print Time Genius, which i'm not using. Instead, i'm using this plugin. It makes sense that Print Time Genius is used for estimation if your slicer doesn't output M73 commands, but mine does, and i like using this more simple plugin instead. It would be nice if this plugin had an option to show the time (maybe just in minutes, doesn't have to be super precise) until the next tool chance. If it simply displayed the difference between the last M73 command sent and the last M73 command before the next tool change, it would be precise enough i think.

NilsRo, 2022-04-26:
Should be easy to add as I still inject code in the objects used for that. Code also implement an percentage rule but would be better if derived from the code. 

Will inform you when a first Beta is available...

codingcatgirl, 2022-04-26:
Awesome, thank you very much!

NilsRo, 2022-05-07:
@codingcatgirl   Do you know how MMUs will be handled in GCODE? Could find only some proprietary codes for Prusa like T1,...

codingcatgirl, 2022-05-07:
T1 is not a proprietary code :) It's a totally normal tool change, and it's the printer firmware that translates commands like T1, T2 into the appropriate commands to send to the MMU. at least for the one i use.

Sorry, i'll test it as soon as i have my nozzle unclogged again :D

NilsRo, 2022-05-07:
Ok, good to know. Find it also now on marlinfw.org. Was hidden by the disclaimer I did not accepted. :-) Will add it also beside M600.

codingcatgirl, 2022-05-07:
It's easy to miss all the way on the bottom, yes :) Of course commands can go further than T6 for some firmwares and setups

NilsRo, 2022-05-08:
Both is available now in filebrowser and also during print.

[image]

codingcatgirl, 2022-05-08:
That looks great, however in a multi material print you might have dozens of filament changes, will it list them all? that might become a bit too much quickly.

NilsRo, 2022-05-08:
Actually yes. How behave OctoPrint? It still shows the tool switch and filament length.

Could you upload a realistic example than I can test a little bit. And also play around with it on your own. It cannot crash a print even if an error occurs.

codingcatgirl, 2022-05-08:
Here's [this om nom](https://www.thingiverse.com/thing:2512972) sliced with the default Prusa MMU2S profiles. 188 Tool changes ^^;

[OmNom_White_V2_0.2mm_PLA_MK3SMMU2S_7h29m.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/8646734/OmNom_White_V2_0.2mm_PLA_MK3SMMU2S_7h29m.zip)

codingcatgirl, 2022-05-08:
In a way it's not a perfect example because with this many tool changes you'd just rely on it working well anyway… but in a way it's also a good example because the first tool change only happens 63 layers in so it's totally useful to know when that will be and how much time you have to leave your printer alone if you wanna be there for the tool changes.

NilsRo, 2022-05-08:
Decided to show all changes in the filebrowser but inly the next 10 in the printer view.

[image]

NilsRo, 2022-05-08:
By the way...do you know my Metadata enhancement and is it possible to use with PrusaSlicer...using Cura only...

codingcatgirl, 2022-05-08:
Oh that sounds like a good idea. I don't know of the Metadata enhancement, but looking at ti it should be possible to configure the slicer accordingly. Tbh i'm using SuperSlicer, which is a prusaslicer fork. But i think at the end of the gcode the slicer settings should be findable anyway, with the filament for each material? You could implement that, or get the info from the slicer setting octoprint plugin that already extracts them from the gcode.

But adding comments to the gcdoe can be done using custom filament change gcode in superslicer and probably prusaslicer too. i can look up hwo to do it in the right format and post about it here (if i think about it, feel free to remind me)

codingcatgirl, 2022-05-08:
Ok, since i just had a minute anyway. IN the gocde i sent, in the end you can find the usual Slic3r/PrusaSlicer/Superslicer output format: 
 
```
; filament_colour = #FF8000;#FF8000;#FF8000;#FF8000;#FF8000
; filament_cooling_final_speed = 2,2,2,2,2
; filament_cooling_initial_speed = 3,3,3,3,3
; filament_cooling_moves = 1,1,1,1,1
; filament_cost = 30.24,30.24,30.24,30.24,30.24
; filament_density = 1.24,1.24,1.24,1.24,1.24
; filament_diameter = 1.75,1.75,1.75,1.75,1.75
; filament_load_time = 15,15,15,15,15
; filament_loading_speed = 14,14,14,14,14
; filament_loading_speed_start = 19,19,19,19,19
; filament_max_volumetric_speed = 13,13,13,13,13
; filament_minimal_purge_on_wipe_tower = 15,15,15,15,15
; filament_notes = ;;;;
; filament_ramming_parameters = "130 120 2.70968 2.93548 3.32258 3.83871 4.58065 5.54839 6.51613 7.35484 7.93548 8.16129| 0.05 2.66451 0.45 3.05805 0.95 4.05807 1.45 5.97742 1.95 7.69999 2.45 8.1936 2.95 11.342 3.45 11.4065 3.95 7.6 4.45 7.6 4.95 7.6";"130 120 2.70968 2.93548 3.32258 3.83871 4.58065 5.54839 6.51613 7.35484 7.93548 8.16129| 0.05 2.66451 0.45 3.05805 0.95 4.05807 1.45 5.97742 1.95 7.69999 2.45 8.1936 2.95 11.342 3.45 11.4065 3.95 7.6 4.45 7.6 4.95 7.6";"130 120 2.70968 2.93548 3.32258 3.83871 4.58065 5.54839 6.51613 7.35484 7.93548 8.16129| 0.05 2.66451 0.45 3.05805 0.95 4.05807 1.45 5.97742 1.95 7.69999 2.45 8.1936 2.95 11.342 3.45 11.4065 3.95 7.6 4.45 7.6 4.95 7.6";"130 120 2.70968 2.93548 3.32258 3.83871 4.58065 5.54839 6.51613 7.35484 7.93548 8.16129| 0.05 2.66451 0.45 3.05805 0.95 4.05807 1.45 5.97742 1.95 7.69999 2.45 8.1936 2.95 11.342 3.45 11.4065 3.95 7.6 4.45 7.6 4.95 7.6";"130 120 2.70968 2.93548 3.32258 3.83871 4.58065 5.54839 6.51613 7.35484 7.93548 8.16129| 0.05 2.66451 0.45 3.05805 0.95 4.05807 1.45 5.97742 1.95 7.69999 2.45 8.1936 2.95 11.342 3.45 11.4065 3.95 7.6 4.45 7.6 4.95 7.6"
; filament_settings_id = "Prusament PLA @MMU2";"Prusament PLA @MMU2";"Prusament PLA @MMU2";"Prusament PLA @MMU2";"Prusament PLA @MMU2"
; filament_soluble = 0,0,0,0,0
; filament_spool_weight = 201,201,201,201,201
; filament_toolchange_delay = 0,0,0,0,0
; filament_type = PLA;PLA;PLA;PLA;PLA
; filament_unload_time = 12,12,12,12,12
; filament_unloading_speed = 20,20,20,20,20
; filament_unloading_speed_start = 100,100,100,100,100
; filament_vendor = Prusa Polymers
```

Filament UUIDs aren't really a thing there, though you can add custom filament variables.

codingcatgirl, 2022-05-08:
Since i just looked it up, if one adds this to the Tool Change G-Code (all confirmed to work in both prusaslicer and and superslicer:

```
;Filament Settings ID: {filament_settings_id[next_extruder]}
;Filament Colour: {filament_colour[next_extruder]}
;Filament Notes: {filament_notes[next_extruder]}
;Filament Cost: {filament_cost[next_extruder]}
;Filament Type: {filament_type[next_extruder]}
;Filament Soluble: {filament_soluble[next_extruder]}
```

you get this before the tool change in question: 

```
;Filament Settings ID: Prusament PLA @MMU2
;Filament Colour: #FF8000
;Filament Notes: testnote
;Filament Cost: 30.24
;Filament Type: PLA
;Filament Soluble: false
```

custom filament variables are only a thing in superslicer, not prusaslicer. notes is a free text field

it's probably a better idea to extract it from the comments at the end

NilsRo, 2022-05-09:
So it's possible to add it in the format required by the plugin? I do not like to pass all comments as there are many that are not usable which creates many entries in the settings. That's are the thoughts behind. But if that is not possible I can add a new configuration page.

Regarding the tool changes, is that fine from your point of view?

codingcatgirl, 2022-05-10:
I'm not sure if it's possible to add it perfectly like required by the plugin. Since you do require UUIDs, right? Can't you just install prusaslicer and try it out?

NilsRo, 2022-05-10:
I need it in (;Slicer info:<key>;<value>)
Actually it is (; <key> = <value>)

As there are many comments sometimes I like to get a sign like ;Slicer info to get it marked for the plugin. But if it is difficult I can add an regex but it would be not that comfortable.

And some comments to the plugin? If it is fine I like to release it soon that it is available for everybody.

codingcatgirl, 2022-05-10:
*nods* the point is with superslicer and prusaslicer you can use templates to output any comments you want in your gcode so if you provide the snipped you're supposed to insert in the settings it can display pretty much anything

NilsRo, 2022-05-11:
Perfect, good to know not to change something unnecessary. New version is released...please disable the development branch...

closed 2022-05-11 by NilsRo