NilsRo/OctoPrint-SlicerEstimator #29: M600 time estimate?

WimVDK, 2022-05-14:
I've been tinkering with my first color changing prints. Using Cura and the filament change plugin (M600 option). I print exclusively via octtoprint and found this plugin initially. https://plugins.octoprint.org/plugins/TimeToFilament/ It worked a wopping 1x in detecting the M600 and giving me a time estimate (and it was off by over 3hrs). (and hasn't seen an update in over 15 months) After doing a lot more searching i found yours.

Now according to the Tool & filament changes section images, M600 is supported for time estimate, but then under "notes" it says 3

"Filament change time is supported with M117 or M73 commands only as actually it is not possible to derive the correct progress the M600 command is found in the GCODE."

What i know about M117 is that it's used to send messages to the printer, but that also spams the notification section in the octoprint UI

So i guess my question is how can i use the plugin and get an accurate M600 filament change estimate without spamming my octoprint notification section with time updates. I'm currently PrintTimeGenius and overall print estimates are pretty accurate, but i don't mind switching if i have too.

NilsRo, 2022-05-15:
Hi. 

I am working on using inline comments that are added with newer Cura versions but they are not published by GCODE sent event in OctoPrint. I have to investigate how to get this comments and do a refactoring of the code then...that needs some time
...in the meanwhile you can use M117 and perhaps disable notifications in Cura. You can do that in the plugin section. 

Would be also interesting to get more feedback about the M600 change function as I am creates functional models without filament change only.

WimVDK, 2022-05-15:
Please elaborate on "newer versions" i'm using the latest stable releases of both octoprint (1.7.3) and cura (4.13.1). I'd be happy to assist/test. any chance you can tell me exactly what i need to enable/configure to get

1. M600 being detected reliably
2. accurate M600 time estimate

NilsRo, 2022-05-15:
The necessary GCODE comments seems to be added with Cura 4 so you are fine, but at first I have to refactor the plugins code.

Actually it would be nice if you can test the M117 implementation. Its only some days old and the base for future enhancements. 

I will inform you if an alpha is available supporting the Cura comments, would enjoy if you can test this then also.

WimVDK, 2022-05-15:
I'd be happy if you can tell me what i need to do cura wise :)

NilsRo, 2022-05-15:
Simply do the post processing described here: https://github.com/NilsRo/OctoPrint-SlicerEstimator#cura-m117
And in Slicer Estimator Settings disable Slicer autodetection and manually set Cura M117.

Everything should be fine then.

NilsRo, 2022-08-07:
Reworked everything and also added a Cura native handling. So dropped M117 handling as it is obsolete. Please update to 1.5.0 released now. For M600 handling supported now for every slicer you have to upload a file again to OctoPrint as there are commands added to the GCODE file.

closed 2022-08-07 by NilsRo