NilsRo/OctoPrint-SlicerEstimator #36: SlicerEstimator no longer pulling time notes out of prusaslicer files

SamD98, 2022-08-23:
Slicer Estimator is no longer pulling time out of the slicer M73 commands from the exact files it used to.  Files were not resliced or reuploaded but time estimation has stopped working.

**Plugins installed**
- Bed Cooldown[bedcooldown] v0.4.1
- Camera Settings[camerasettings] v0.4.0
- Exclude Region[excluderegion] v0.3.0
- ExtraFileInfo[extrafileinfo] v1.3.0
- Filament Manager[filamentmanager] v1.9.1
- Floating Navbar[floatingnavbar] v0.3.7
- GcodeEditor[GcodeEditor] v0.2.12
- GPIO Control[gpiocontrol] v1.0.8
- Navbar Temperature Plugin[navbartemp] v0.15
- Octolapse[octolapse] v0.4.1
- PrettyGCode[prettygcode] v1.2.4
- Prusa MMU2 Select Filament[mmu2filamentselect] v1.0.7
- Slicer Estimator[SlicerEstimator] v1.5.2
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- SlicerSettingsParser[SlicerSettingsParser] v2.0.1
- UI Customizer[uicustomizer] v0.1.9.0

**Settings**
- average_prio: true
- metadata_filelist: true
- metadata_filelist_align: top
- metadata_list: 
- metadata_printer: true
- pdp: 1
- php: 1
Installed plugins: (SlicerEstimator)
- pmp: 2
- psp: 3
- pwp: 1
- slicer: 1
- useDevChannel: false
- use_assets: true

**Software versions**
- OctoPrint 1.8.2
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.0.0 Safari/537.36

NilsRo, 2022-08-23:
The handling has to be changed that files has to be uploaded again to add the printing time to the file itself. But if you still printed it the last real print time is used by OctoPrint/Plugin according your settings which is much better than an estimation as it is the real print time. 

In the actual version you should get a short notification that there is no print time available and you have to upload the file again. This you can ignore as you have average estimation enabled as fallback. You will see a green dot then.

On upload files there is a bug, I forgot to delete a debugging setting for the notifications so it will be shown everytime on upload. This is solved in 1.5.3 I will publish now.

adcurtin, 2022-08-27:
even with the latest version, it doesn't seem to read the gcodes out of the file.

the prusaslicer estimate is very accurate, I don't need any other estimate :)

M73 P1 R86
M73 Q1 S86 

are the gcodes in my current print that aren't getting interpreted.

NilsRo, 2022-08-27:
> even with the latest version, it doesn't seem to read the gcodes out of the file.
> 
> 
> 
> the prusaslicer estimate is very accurate, I don't need any other estimate :)
> 
> 
> 
> M73 P1 R86
> 
> M73 Q1 S86 
> 
> 
> 
> are the gcodes in my current print that aren't getting interpreted.

Can you upload a GCODE file. My prusa test gcode goes through fine but perhaps there is something different with yours.

adcurtin, 2022-08-27:
[ide hard drive mount t2_0.2mm_PETG_MK3S_1h27m.gcode.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/9437457/ide.hard.drive.mount.t2_0.2mm_PETG_MK3S_1h27m.gcode.zip)

NilsRo, 2022-08-27:
@adcurtin : Please enable the development version in the settings of the plugin, update to version 1.5.4 and test again...

adcurtin, 2022-08-27:
it's fixed, working perfectly now 

(commit 7c3eddf14a66f18229c71021ecf259fe4e76c4fd is what I've got)

NilsRo, 2022-08-28:
Published 1.5.4 now. Please disable the dev-mode to avoid issues. Perhaps you have to force the update in Software updater in the drop down.
[image]

closed 2022-08-28 by NilsRo

adcurtin, 2022-08-28:
I did not need to force an update, it found 1.5.4 automatically after disabling dev mode.

thanks for the quick fix!