NilsRo/OctoPrint-SlicerEstimator #37: Slicer not Detected

Gozania, 2022-08-23:
Slicer Not Dectected
When I add newly sliced files (Prusaslicer 2.42 M73 enabled)
Describe your problem?  See photos
What is the problem? See Photos
Can you recreate it? Yes
Did you try disabling plugins? No
Did you remember to update the subject?**

**Plugins installed**
- Themeify[themeify] v1.2.2
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.0
- Preheat Button[preheat] v0.8.0
- TP-Link Smartplug[tplinksmartplug] v1.0.2
- Slicer Estimator[SlicerEstimator] v1.5.2
- Resource Monitor[resource_monitor] v0.3.6
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8
- PrettyGCode[prettygcode] v1.2.4
- Plotly Temp Graph[plotlytempgraph] v0.1.3
- OctoPod Plugin[octopod] v0.3.14
- Obico for OctoPrint[obico] v2.1.2
- Fullscreen Plugin[fullscreen] v0.0.6
- Filament Manager[filamentmanager] v1.9.1
- Dashboard[dashboard] v1.19.8
- Camera Settings[camerasettings] v0.4.0
- BLTouch Plugin[BLTouch] v0.3.4
- Octolapse[octolapse] v0.4.1
- OctoDash Companion[octodashcompanion] v0.0.7
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Bed Visualizer[bedlevelvisualizer] v1.1.1

**Settings**
- average_prio: true
- metadata_filelist: false
- metadata_filelist_align: top
- metadata_list: 
- metadata_printer: true
- pdp: 1
- php: 1
Installed plugins: (SlicerEstimator)
- pmp: 2
- psp: 3
- pwp: 1
- slicer: 1
- useDevChannel: false
- use_assets: true

**Software versions**
- OctoPrint 1.8.2
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:103.0) Gecko/20100101 Firefox/103.
[image]

[image]

NilsRo, 2022-08-23:
Solved in 1.5.3. Forgot to delete a debugging setting to test the notifications so it is raised every time. You can ignore that until you did not updated the plugin it is a false warning only.  Will publish the new version now.

closed 2022-08-23 by NilsRo

Gozania, 2022-08-23:
Thanks! Will update after print is done.