NilsRo/OctoPrint-SlicerEstimator #40: Slicer Not Detected Message

proxycg, 2022-08-26:
## Description
**ENTER DESCRIPTION HERE
Describe your problem? Slicer Not Detected Message
What is the problem? Slicer Not Detected Message
Can you recreate it? Yes, whenever I start a print
Did you try disabling plugins? The required plugins are the ones I have installed.
Did you remember to update the subject?**

**Plugins installed**
- Cost Estimation[costestimation] v3.5.0 - YES
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0 - YES
- Octolapse[octolapse] v0.4.1 - YES
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.8 - YES
- Slicer Estimator[SlicerEstimator] v1.5.3 - YES

**Settings**
- average_prio: true
- metadata_filelist: true
- metadata_filelist_align: top
- metadata_list: 
- metadata_printer: true
Installed plugins: (SlicerEstimator)
- useDevChannel: false
- use_assets: true

**Software versions**
- OctoPrint 1.8.2
- Python 3.10.6

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.5112.102 Safari/537.36 Edg/104.0.1293.63

NilsRo, 2022-08-26:
Hi,

can you upload a file that throws that warning? What slicer are you using?

Cheers,
Nils

proxycg, 2022-08-26:
Hi Nils,

I uploaded the calibration file generated from SuperSlicer. No matter what I upload I always get the message, with all the slicers. I am relatively new to the 3D Printing world, so hopefully it is not a user error in the configuration.

[Flow calibration_25m_0.20}mm_220C_PLA_ENDER3S1.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/9435403/Flow.calibration_25m_0.20.mm_220C_PLA_ENDER3S1.zip)

proxycg, 2022-08-26:
Here are screen captures of the three tabs.

[image]
[image]
[image]
.

NilsRo, 2022-08-26:
There are 2 problems.

1. SuperSlicer is not supported yet
2. The remaining time calculation is deactivated in your example. Even if I add support for SuperSlicer there are not time codes included in the file. 

I can add SuperSlicer support but I need an example file with time codes into it. If you can upload one it would be not that complex. Changed the structure of the plugin with 1.5 to simplify this task.

proxycg, 2022-08-26:
Thank you. I have some questions.

Is it deactivated because I missed a setting?

How do I insert the timecodes? I don't know the process.

NilsRo, 2022-08-26:
As Super Slicer is a fork of Prusa Slicer this could work.

https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-general-discussion-announcements-and-releases/time-remaining-estimate/

proxycg, 2022-08-27:
Thank you, I'm not sure this will work for me, but I do need to do more research to make sure everything is setup properly.

If you can add support for SuperSlicer, that would be great. I would provide the file you requested, but I do not know how to do this. If you provide instructions, or a link to instructions, I will be happy to contribute and create the files necessary.

NilsRo, 2022-08-27:
Hi need a Gcode file with estimation active. But I think the Gcodes for time estimation are the same with Prusa Slicer but I am not sure.

NilsRo, 2022-09-05:
Super Slicer is supported now.

closed 2022-09-05 by NilsRo

proxycg, 2022-09-05:
Thank you! You're the best.