NilsRo/OctoPrint-SlicerEstimator #41: slicer estimator no slicer detected

biketech2727, 2022-08-27:
## Description
**ENTER DESCRIPTION HERE
I have a pop up saying no slicer detected. It had worked before since an update a few weeks back. Im using Super Slicer version 2.4.58.4
Describe your problem?
What is the problem? pop up says to put in a ticket if slicer should be known - Super Slicer 2.4.58.4
Can you recreate it? ever time i slice and upload to Octoprint
Did you try disabling plugins? no
Did you remember to update the subject?** plugin has latest update per octoprint plugins - all plugins are current

**Plugins installed**
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Calibration-Companion[calibrationcompanion] v1.0.1
- Consolidate Temp Control[consolidate_temp_control] v0.1.9
- Dashboard[dashboard] v1.19.8
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- DisplayZ[displayz] v0.2.0
- Draggable Files[draggable_files] v1.1.2
- FileManager[filemanager] v0.1.6
- ipOnConnect[ipOnConnect] v0.2.4
- Macro[macro] v0.3.2
- OctoDash Companion[octodashcompanion] v0.0.7
- Octolapse[octolapse] v0.4.1
- PrettyGCode[prettygcode] v1.2.4
- Printer Stats[stats] v3.1.3
- Printoid Plugin[printoid] v1.2.1
- Simple Filament Change Buttons[simple_filament_change_buttons] v1.1
- Slicer Estimator[SlicerEstimator] v1.5.3
- STL Viewer[stlviewer] v0.4.2
- UI Customizer[uicustomizer] v0.1.9.0
- WebcamTab[webcamtab] v0.3.0
- Z Probe Offset[z_probe_offset] v0.5

**Settings**
- average_prio: true
- metadata_filelist: true
- metadata_filelist_align: top
- metadata_list: 
- metadata_printer: true
- pdp: 1
- php: 1
Installed plugins: (SlicerEstimator)
- pmp: 2
- psp: 3
- pwp: 1
- useDevChannel: false
- use_assets: true

**Software versions**
- OctoPrint 1.8.2
- Python 3.8.10

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.0.0 Safari/537.36

NilsRo, 2022-08-27:
Super Slicer was not directly supported but worked. Because of some simplifications that is not the case anymore. But I added Support in the actual dev version. Simply activate it under development tab in the plugins settings and report back if it is fine with version 1.5.4.

NilsRo, 2022-08-28:
Published 1.5.4 now you can also try there without changing to dev mode. Would be happy to get an update if everything is fine...

closed 2022-09-05 by NilsRo