NilsRo/OctoPrint-SlicerEstimator #47: Unable to determine slicer

Mubo507, 2022-09-26:
## Description

**Using creality slicer for my ender 3 v2. Worked fine for a while then the estimator stopped working.

Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
What slicer are you using?
Did you uploaded the GCODE file causing the issue?
Did you remember to update the subject?**

**Plugins installed**
- Bed Leveling Wizard[bedlevelingwizard] v0.2.4
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- Obico for OctoPrint[obico] v2.2.1
- OctoEverywhere![octoeverywhere] v1.10.5
- Octolapse[octolapse] v0.4.1
- Slicer Estimator[SlicerEstimator] v1.5.5

**Settings**
- average_prio: true
- metadata_filelist: true
- metadata_filelist_align: top
- metadata_list: 
- metadata_printer: true
- metadata_slicer: true
- pdp: 1
- php: 1
Installed plugins: (SlicerEstimator)
- pmp: 2
- psp: 3
- pwp: 1
- useDevChannel: false
- use_assets: true

**Software versions**
- OctoPrint 1.8.3
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/105.0.0.0 Safari/537.36

NilsRo, 2022-09-26:
Creality Slicer is not supported. If you provide a file I can check if it makes sense to integrate it. 

Newest version of plugin shows a warning something is wrong or not detected.

Mubo507, 2022-09-26:
It's always recognized it as Cura, not sure why it stopped. 
Here is a recent gcode file
[CE3V2_WasherDogs2.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/9646545/CE3V2_WasherDogs2.zip)

NilsRo, 2022-09-26:
If you enable development mode in the settings of the plugin you will update to a version that should support Creality Slicer. Would be nice to get a short feedback if it is fine.

Creality Slicer is based on Cura so I have to add it to the detection and also that it worked before without auto detection...

NilsRo, 2022-10-05:
@Mubo507 : Did you tested the change? Is it working fine on your site?

closed 2022-10-07 by NilsRo