NilsRo/OctoPrint-SlicerEstimator #48: Any way to support parsing of files at print time, not upload time?

cmlad, 2022-10-06:
I sync my Gcode files via Syncthing to OctoPrint, rather than uploading through the UI. Is there a way to make Slicer Estimator optionally do the estimation at print start? Or have a background process that looks at all Gcode files? Or maybe use the plugin functionality?

I can make a PR if you are ok with this.

NilsRo, 2022-10-06:
Are you doing this with help of a OctoPrint Plugin? I attached to the official API so then it should be changed in the sync plugin. Or do 'copy' it to a folder with that sync?

The files are changed during API call so I have to check if it is supported officially to change it later.

cmlad, 2022-10-06:
Thanks for the fast response.

I use SyncThing directly as I couldn't find a plugin for it. It just copies files into the upload folder of OctoPrint.

Let me look into this on my own more as well - I didn't realize files are changed, but I ran a test and see what happens. I can probably write a plugin that monitors files and simulates re-uploading if there is no metadata.

NilsRo, 2022-10-07:
You can take a look here: https://community.octoprint.org/t/what-folder-holds-uploaded-files/6517/6

cmlad, 2022-10-07:
Thanks, that's a great pointer. Let me close and I will figure it out.

closed 2022-10-07 by cmlad