NilsRo/OctoPrint-SlicerEstimator #50: No print time estimation from slicer available. Please upload GCODE file again.

MarshMalloW0007, 2022-10-21:
_**Simplify3D**

With Simplify3D no changes has to be applied to Simplify3D. The overall print time is read out of a comment in the GCODE. For a correct estimation OctoPrints percentage done is used as there is only the overall print time available._

**At the end of my gcode file :**
;
; Build Summary
;   Build time: 22 hours 11 minutes
;   Filament length: 33476.6 mm (33.48 m)
;   Plastic volume: 80520.55 mm^3 (80.52 cc)
;   Plastic weight: 100.65 g (0.22 lb)
;   Material cost: 3.02

NilsRo, 2022-10-22:
Can you upload a gcode file that I can reproduce it. I am on vacation and will check it when I am back.

MarshMalloW0007, 2022-10-22:
[TT-S3D-SupportCasqueAudioE_Arceau-1P-PPLAPlus_A13_04_10s_215.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/9845276/TT-S3D-SupportCasqueAudioE_Arceau-1P-PPLAPlus_A13_04_10s_215.zip)

NilsRo, 2022-10-28:
@MarshMalloW0007 : You can activate dev mode in the settings and update to the newest dev version. I implement a fix for that. 

BTW. The remaining time in your uploaded example it different from your first post. There was a bug in the regex that "1 minute" was not recognized but 22 hours 11 minutes should work fine. So please test both. - It could be also the case that your Github plugin does not use the official API for file management so the scan of the GCODE would not be done. This would then be the case for all plugins related on this and a bug in that plugin. But let us see how your tests are going...

MarshMalloW0007, 2022-10-28:
Hello NilsRo,

I'm sorry, I posted another file with the same problem.

I will try the dev version.

NilsRo, 2022-10-28:
Ok, then perhaps we have a second problem. But let us see if the uploaded file will work fine now.

MarshMalloW0007, 2022-10-30:
Hello NilsRo,
The same thing....

closed 2022-11-30 by NilsRo

ModischFabrications, 2022-12-08:
I get the same error, seems like it's not resolved? 

- Uploaded via PrusaSlicer v2.5.0
- OctoPrint 1.8.6
- Slicer Estimator (1.5.7)

NilsRo, 2022-12-10:
> I get the same error, seems like it's not resolved?
> 
> * Uploaded via PrusaSlicer v2.5.0
> * OctoPrint 1.8.6
> * Slicer Estimator (1.5.7)

Hi.

Could you create a new issue and upload your GCODE file. Its not technically a real error its simply that the estimation could not be read out e.g. if it is not added by the slicer or perhaps the format changed. With the GCODE file I can check it and adapt the detection if neccessary. PRusa Slicer is supported so we have to see what caused this.

Cheers,
Nils

MarshMalloW0007, 2022-12-10:
Hello Nils, 
Hello Robin,

Sorry for the late reply.

I took the time to install OctoPrint on another Rapsberry : OctoPrint 1.8.6 Python 3.9.2 OctoPi 1.0.0 Slicer Estimator 1.5.7

Everything works.  Look at the screenshots.

As I had some problems with my production Raspberry/OctoPrint, I reinstalled the Raspberry: OctoPrint 1.8.6 Python 3.7.3 OctoPi 0.18.0.
I reinstalled a backup of my installation. 
The version of Slicer Estimator is 1.5.7
The other problems are gone, but still the same error with Slicer Estimator. 
It is also impossible to do "Refresh Metadata".

I tested with the same file.

[image]
[image]
[image]
[image]
[TT-S3D-Pumpkin_Lid_150-1P-P13_04_10.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/10199909/TT-S3D-Pumpkin_Lid_150-1P-P13_04_10.zip)

NilsRo, 2022-12-11:
@MarshMalloW0007 : The file you uploaded does not contains any metadata tags readable by slicer estimator. It have to looks like described here https://github.com/NilsRo/OctoPrint-SlicerEstimator#metadata

NilsRo, 2022-12-11:
Forget it: I still implemented it. Haha. Will have a look...

MarshMalloW0007, 2022-12-11:
Like this : 

;Slicer info:material_brand;Sunlu
;Slicer info:material_name;Yellow/133051
;Slicer info:temperatureSetpointTemperatures;200,195,55,50

at the bottom file => same result.

[TT-S3D-Pumpkin_Lid_150-1P-P13_04_10.01.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/10201773/TT-S3D-Pumpkin_Lid_150-1P-P13_04_10.01.zip)

It's work in my dev OctoPrint.
I think it's a conflict with another plugin.

NilsRo, 2022-12-11:
The standard comments are still parsed so the special comments are necessary only if you like to add additional infos. I found a bug which is perhaps related but I am not sure. It breaks the JS GUI if you are logged in during upload sometimes...timing issue...I fixed it in the actual development branch. But to work around it you have to refresh the browser. 

Give it a try, you can change to dev version in the plugin settings. 

If that does not fix the issue please enable the debug log for Slicer Estimator that we can see some additional info in the logs and add the log but please open a new issue. You can open it in from the settings and it will add your installed plugins automatically.  

This issue was opened because of the remaining print time that was sometimes not detected.

MarshMalloW0007, 2022-12-11:
I found how to make the plug-in work, but not with the way I work.

If I upload the file with the browser, it works, I have access to the metadata, the Refresh MetaData button.

If I upload the file in SSH or via my GIT, it doesn't work.

NilsRo, 2022-12-11:
@MarshMalloW0007  Then you are not using the official API with that workflow. There is a upload folder where the files has to be placed that it will be scanned by OctoPrint. That is a system immanent design.

MarshMalloW0007, 2022-12-11:
My files are uploaded in ssh/git in the folder defined in the OctoPrint configuration. 

However, I don't know if OctoPrint imposes to upload only via the browser.

[image]
[image]

NilsRo, 2022-12-11:
https://community.octoprint.org/t/how-to-add-files-directly-to-upload-directory/19704

Its not allowed to upload to upload folder it has to be managed by OctoPrint. You have to use the watched folder.

MarshMalloW0007, 2022-12-11:
Thank you,
I'm reading this and looking at what I need to change in my flow.

ModischFabrications, 2022-12-11:
Minor addition to my error: "Printer Settings > Supports remaining times" was unchecked, this might have been the error.

MarshMalloW0007, 2022-12-12:
Hello Nils,

Everything works, thanks for your help.

Before, I was using the OctoPrint-GitFiles plug-in and I didn't care about the watched folder....
Then, when the plug-in became obsolete (python <3), I managed the pull manually and completely forgot about the need for the watched folder.

The problem was not where it was...