NilsRo/OctoPrint-SlicerEstimator #53: Slicer Estimator has an issue  with Simplify3d Version 5

rahatch1, 2022-12-16:
## Description

**ENTER DESCRIPTION HERE

Describe your problem? After Upgrading to Simplify3d Version 5, Started getting Error message when trying to print any GCODE file created with the new Simplify3d Version 5.

What is the problem? Two popup error messages when stating a new print that was sliced with Simplifiy3d Version5.
First Error message:  
     Slicer Estimator
     No timecodes found. Please check if the remaining time feature in the slicer is active

Second Error message: 
     Slicer Estimator
      No print time estimation from slicer
      available. Please upload GCODE file again.
      The file was uploaded before Slicer Estimator Was installed or the Slicer was not detected.

Note.  The Slicer Estimator was installed prior to the file being uploaded.

Can you recreate it? Yes
Did you try disabling plugins? No
What slicer are you using? Simplify3d Version 5
Did you uploaded the GCODE file causing the issue? No,  I can if you need it.
Did you remember to update the subject? Yes

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Cancel Objects[cancelobject] v0.4.7
- Change Filament Plugin[Change_Filament] v0.3.2
- Cost Estimation[costestimation] v3.5.0
- Dashboard[dashboard] v1.19.9
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- DisplayProgress[displayprogress] v0.1.3
- FileManager[filemanager] v0.1.6
- Firmware Updater[firmwareupdater] v1.13.3
- Gcode Leveling[gcodeleveling] v0.4.0
- InlineConfirm[inlineconfirm] v1.1.4
- Marlin EEPROM Editor[eeprom_marlin] v3.2.0
- Navbar Temperature Plugin[navbartemp] v0.15
- Preheat Button[preheat] v0.8.0
- Resource Monitor[resource_monitor] v0.3.9
- Restore Leveling After G28[restorelevelingafterg28] v1.1.0
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Simple Filament Change Buttons[simple_filament_change_buttons] v1.1
- Slicer Estimator[SlicerEstimator] v1.5.8
- Terminal Commands Extended[terminalcommandsextended] v0.1.7
- Z Probe Offset[z_probe_offset] v0.5
- ΔµCalibrator[DuCalibrator] v1.0.4

**Settings**
- average_prio: true
- metadata_filelist: true
- metadata_filelist_align: top
- metadata_list: 
- metadata_printer: true
- metadata_slicer: true
- pdp: 1
- php: 1
Installed plugins: (SlicerEstimator)
- pmp: 2
- psp: 3
- pwp: 1
- slicer: 3
- useDevChannel: false
- use_assets: true

**Software versions**
- OctoPrint 1.8.6
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36

NilsRo, 2022-12-16:
Hi. Can you upload a GCODE example. Perhaps the format changed. Using Cura myself so not aware of changes of other slicers...

Cheers
Nils

rahatch1, 2022-12-17:
1

NilsRo, 2022-12-17:
Can you upload the file as an attachment. Looks cut at the end.

rahatch1, 2022-12-18:
First time trying to attach a file, ended up having to zip the gcode file.

[Created with Simplify3d-v5.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/10253002/Created.with.Simplify3d-v5.zip)

NilsRo, 2022-12-18:
Problem solved...the format was changed in Simplify3D V5. I released a patch that will be offered in your OctoPrint instance the next days. If you like to get it earlier you can force the update.

Cheers,
Nils

closed 2022-12-18 by NilsRo