NilsRo/OctoPrint-SlicerEstimator #55: SlicerEstimator makes /api/files/local response 3x bigger

the-ress, 2023-01-03:
SlicerEstimator adds a large `slicer_metadata` field to each file and it makes `/api/files/local` response 3x bigger causing performance issues. And I don't think OctoPrint has a way to get a file list without these extra data.

I have about 500 gcode files and each has ~275 entries in `slicer_metadata` (I'm using PrusaSlicer). When listing the files on RPi Zero 2, it spends over 6 seconds just serializing to JSON.

NilsRo, 2023-01-04:
Could you provide the result of the API call or the metadata entries causing this. Also a single GCODE file would be nice to have an example. 

I like to check if there are some single larger elements. Also you can disable storing the standard metadata in the plugins settings and define your own to reduce the amount entries. With my PI4 I cannot see any noticeable difference...

the-ress, 2023-01-05:
Here's one file as an example:

gcode: [08-Adjust Button_0.1mm_PA_MK3S_14m.txt](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/10356229/08-Adjust.Button_0.1mm_PA_MK3S_14m.txt)

<details>
<summary>relevant part of API response</summary>

```json
        {
            "dashboard": {
                "filament_change_array": "[]",
                "layer_move_array": "[4, 469, 460, 466, 395, 387, 390, 390, 389, 390, 387, 390, 390, 390, 390, 502, 498, 498, 486, 498, 468, 497, 505, 490, 504, 484, 496, 496, 503, 503, 505, 505, 490, 503, 503, 504, 502, 502, 511, 678, 678, 678, 691, 691, 690, 640, 642, 642, 643, 638, 642, 641, 641, 641, 641, 641, 641, 643, 643, 641, 643, 644, 643, 636, 646]"
            },
            "date": 1670882966,
            "display": "08-Adjust Button_0.1mm_PA_MK3S_14m.gcode",
            "gcodeAnalysis": {
                "dimensions": {
                    "depth": 116.425,
                    "height": 6.5,
                    "width": 133.425
                },
                "estimatedPrintTime": 840.0,
                "filament": {
                    "tool0": {
                        "length": 171.59231000001026,
                        "volume": 0.4127278732018468
                    }
                },
                "printingArea": {
                    "maxX": 133.425,
                    "maxY": 113.425,
                    "maxZ": 6.5,
                    "minX": 0.0,
                    "minY": -3.0,
                    "minZ": 0.0
                }
            },
            "hash": "57057b70d4481575e812f0a548205cda703be935",
            "name": "08-Adjust Button_0.1mm_PA_MK3S_14m.gcode",
            "origin": "local",
            "path": "08-Adjust Button_0.1mm_PA_MK3S_14m.gcode",
            "prints": {
                "failure": 0,
                "last": {
                    "date": 1670896387.579137,
                    "printTime": 873.8343942000065,
                    "success": true
                },
                "success": 1
            },
            "refs": {
                "download": "http://127.0.0.1:8080/downloads/files/local/08-Adjust%20Button_0.1mm_PA_MK3S_14m.gcode",
                "resource": "http://127.0.0.1:8080/api/files/local/08-Adjust%20Button_0.1mm_PA_MK3S_14m.gcode"
            },
            "size": 1197085,
            "slicer_additional": {
                "bytes": 1192943,
                "lines": 53917,
                "printtime": 840.0,
                "slicer": "1"
            },
            "slicer_filament_change": null,
            "slicer_metadata": {
                "avoid_crossing_perimeters": "0",
                "avoid_crossing_perimeters_max_detour": "0",
                "bed_custom_model": "",
                "bed_custom_texture": "",
                "bed_shape": "0x0,250x0,250x210,0x210",
                "bed_temperature": "115",
                "before_layer_gcode": ";BEFORE_LAYER_CHANGE\\nG92 E0.0\\n;[layer_z]\\n\\n",
[... 261 lines cut ...]
                "wipe_tower_x": "170",
                "wipe_tower_y": "125",
                "wiping_volumes_extruders": "70,70",
                "wiping_volumes_matrix": "0",
                "xy_size_compensation": "0",
                "z_offset": "0"
            },
            "statistics": {
                "averagePrintTime": {
                    "_default": 873.8343942000065
                },
                "lastPrintTime": {
                    "_default": 873.8343942000065
                }
            },
            "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/08-Adjust%20Button_0.1mm_PA_MK3S_14m.png?20221212230935",
            "thumbnail_src": "prusaslicerthumbnails",
            "type": "machinecode",
            "typePath": [
                "machinecode",
                "gcode"
            ]
        }
```
</details>

NilsRo, 2023-01-09:
Needs some time to think about structural changes to perhaps not solve but optimize it. 

In the meanwhile you can use folders as the API will return the files in the actual folder only.

NilsRo, 2023-01-21:
If you switch to development version and update to the newest commit you will see two new buttons in the development tab.

Please press "Update Metadata stored" to update the metadata. Entries larger 50 bytes (like the start gcode) will be ignored now. I like to test if this will speed up the files api or if users have to deselect metadata to reduce the number of entries if the see any issues.

Also it would be nice if you can compare the speed between no custom metadata and metadata published. For that you can delete all metadata "Delete Metadata stored". It can be rebuild with the other button afterwards...

[image]

closed 2023-03-23 by NilsRo