NilsRo/OctoPrint-SlicerEstimator #56: Files created by arc welder are ignored

codingcatgirl, 2023-01-27:
When the plugin "arc welder" is set to create a new file after processing one that was uploaded, this plugin does not seem to analyze it. If the file is printed, ETAs are correctly read from the gocde that is being sent, but before starting the print, no correct total print time estimate is shown and no "time until next toolchange" lines will be visible during the print either.

NilsRo, 2023-01-27:
It seems that this is more or less a problem with arcvelder plugin. From your description octoprints metadata is not copied to the new file so only the time-Codes added directly to the GCODE are available. 

You can leave this open so K can have a look but it first it looks like a missing function in the plugin. 

Should be the same for slicers metadata as they are also stored in OctoPrints file metadata store.

codingcatgirl, 2023-01-27:
Sounds plausible =/ However, stuff like the Slicer Thumbnails plugin seem to pick up on the files perfectly well?

NilsRo, 2023-01-27:
Will check it, have the plug-in installed as well but without that option.

NilsRo, 2023-01-27:
I checked the coding and thumbnail plugin has a special handling in arc welder.
https://github.com/FormerLurker/ArcWelderPlugin/blob/bb71e8fe20cc65f0ab28c0c13685fde39f730320/octoprint_arc_welder/__init__.py#L513-L563

It seems that the file is copied before my plugin finished with setting the metadata as I also see a part to copy the metadata available if I am right. So arc welder should wait until all FILE_ADDED events are processed as before no metadata could be written. Would be nice if you open a bug there.

I can perhaps implement a safety routine that will update the metadata a second time if a file is added to OctoPrint that is not uploaded. But would be better for compatibility that arc welder will wait with all tasks until every plugins finished processing. Perhaps for the the thumbnail handling was introduced...

NilsRo, 2023-01-27:
Ha, think it will work...will provide workaround you can test soon..

NilsRo, 2023-01-27:
Please switch to the development version and try if everything is fine after upload.

codingcatgirl, 2023-01-29:
Tried it, looks like it works. Thank you so much! :)

NilsRo, 2023-01-29:
Will prepare a new release the next days. You should switch back to the standard releases in 1 week or so or if now if you can wait some days for the correction.

Link to OctoPrint issue: https://github.com/OctoPrint/OctoPrint/issues/4719

closed 2023-01-29 by NilsRo