NilsRo/OctoPrint-SlicerEstimator #68: In combination with the plugin for BinaryG-Code from Prusa, the slicer estimator failes

oxivanisher, 2023-11-26:
When using the [OctoPrint-BGCode](https://github.com/jneilliii/OctoPrint-BGCode) plugin, SlicerEstimator failes with a UnicodeDecodeError error:

```plaintext
==> /home/pi/.octoprint/logs/octoprint.log <==
2023-11-24 22:32:44,337 - octoprint.plugins.SlicerEstimator - INFO - Detected PrusaSlicer
2023-11-24 22:32:44,347 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/util/flask.py", line 1613, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/vendor/flask_principal.py", line 196, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/api/files.py", line 671, in uploadGcodeFile
    added_file = fileManager.add_file(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintTimeGenius/__init__.py", line 580, in new_add_file
    return self._file_manager.original_add_file(destination, path, file_object, links, allow_overwrite, printer_profile, None, display)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/__init__.py", line 727, in add_file
    path_in_storage = self._storage(destination).add_file(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/storage.py", line 894, in add_file
    file_object.save(file_path)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/util.py", line 102, in save
    shutil.copyfileobj(source, dest)
  File "/usr/lib/python3.9/shutil.py", line 205, in copyfileobj
    buf = fsrc_read(length)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/util.py", line 211, in read
    line = self.input_stream.readline()
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/util.py", line 230, in readinto
    read = self.read(n)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/util.py", line 214, in read
    processed_line = self.process_line(line)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_SlicerEstimator/metadata.py", line 160, in process_line
    decoded_line = line.decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8a in position 1: invalid start byte
```

This is the corresponding issue in the OctoPrint-BGCode repo: https://github.com/jneilliii/OctoPrint-BGCode/issues/3

jneilliii, 2023-11-26:
@NilsRo pretty sure this was resolved with the preprocessor hook order being set on my plugin. I was able to successfully upload a bgcode without error after making that change.

oxivanisher, 2023-11-26:
This was fixed in the OctoPrint-BGCode plugin. Thanks :)

closed 2023-11-26 by oxivanisher