NilsRo/OctoPrint-SlicerEstimator #71: Super Slicer not detected

CraftoHohenvels, 2024-03-19:
## Description

Addon told me to create an issue...

Problem is the following:
[image]
gcode:
[Benchy.zip](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/14654107/Benchy.zip)

Used Slicer:
SuperSlicer Version 2.5.59
OS: Fedora Workstation 39

**Plugins installed**
- Marlin EEPROM Editor[eeprom_marlin] v3.3.0
- Obico for OctoPrint[obico] v2.4.10
- Octoprint Display ETA[display_eta] v2.2.0
- Queue[queue] v2.0.0
- Firmware Updater[firmwareupdater] v1.14.0
- Slicer Estimator[SlicerEstimator] v1.6.4
- PrettyGCode[prettygcode] v1.2.4
- Themeify[themeify] v1.2.2
- InlineConfirm[inlineconfirm] v1.1.4
- Draggable Files[draggable_files] v1.1.2
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.7
- PSU Control[psucontrol] v1.0.6
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- PSU Control - Fritz!Box[psucontrol_fritz] v1.0.3
- Navbar Temperature Plugin[navbartemp] v0.15
- Preheat Button[preheat] v0.8.0
- Bed Cooldown[bedcooldown] v0.6.0
- Tab Order[taborder] v0.5.12
- Octolapse[octolapse] v0.4.5
- Camera Settings[camerasettings] v0.4.3
- Wattometer[Wattometer] v0.4.0
- Floating Navbar[floatingnavbar] v0.3.7
- Exclude Region[excluderegion] v0.3.2
- GcodeEditor[GcodeEditor] v0.2.14
- Smart Preheat[smartpreheat] v0.0.6
- Autoscroll[autoscroll] v0.0.3

**Settings**
- average_prio: true
- metadata_filelist: true
- metadata_filelist_align: top
- metadata_list: 
- metadata_printer: true
- metadata_slicer: true
Installed plugins: (SlicerEstimator)
- useDevChannel: false
- use_assets: true

**Software versions**
- OctoPrint 1.9.3
- Python 3.10.13

**Browser**
- Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36

NilsRo, 2024-03-19:
Hi,

the GCODE file is processed successfully on my laptop by the plugin. 

How do you upload the file? The processing is done on upload via OctoPrint API. If the file was in OctoPrint before installing the plugin or do not use the common upload methods could lead into missing metadata.

[image]

Cheers,
Nils

CraftoHohenvels, 2024-03-19:
The file was uploaded over Obico

NilsRo, 2024-03-19:
Could you upload the original file directly to OctoPrint via the Webinterface and check if the issue persists? Also if you do not downloaded the file from OctoPrint server could you provide this file. Perhaps Obico is changing the GCODE before uploading... the public API is called of the error is shown so it should not be a problem there.

CraftoHohenvels, 2024-03-20:
The issue persists when uploading directly to the OctoPrint Web-Interface.

NilsRo, 2024-03-20:
Mhh, can you create the debuglog and as a second step disable other plugins. So we can identify if another plugin interfere and enable it step by step. I installed the same OctoPrint and Python version and everything is still fine.

CraftoHohenvels, 2024-03-20:
Update for you:
On a clean OctoPrint installation (same versions as before) I get the same error with the same file.
I do have a snippet of the logs for you:
```
2024-03-20 19:55:08,871 - octoprint.server.preemptive_cache - INFO - ... done in 0.04s
2024-03-20 19:55:15,840 - octoprint.plugins.SlicerEstimator - INFO - Detected SuperSlicer
2024-03-20 19:55:17,152 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Benchy.gcode
2024-03-20 19:55:17,160 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /usr/local/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /octoprint/octoprint/uploads/Benchy.gcode
```

CraftoHohenvels, 2024-03-20:
Debug output of your plugin:
```
2024-03-20 20:02:29,197 - octoprint.plugins.SlicerEstimator - INFO - Detected SuperSlicer
2024-03-20 20:02:29,211 - octoprint.plugins.SlicerEstimator - DEBUG - Metadata line ignored because of it's length.
2024-03-20 20:02:29,212 - octoprint.plugins.SlicerEstimator - DEBUG - Metadata line ignored because of it's length.
2024-03-20 20:02:29,215 - octoprint.plugins.SlicerEstimator - DEBUG - Metadata line ignored because of it's length.
2024-03-20 20:02:29,215 - octoprint.plugins.SlicerEstimator - DEBUG - Metadata line ignored because of it's length.
2024-03-20 20:02:29,221 - octoprint.plugins.SlicerEstimator - DEBUG - Metadata line ignored because of it's length.
2024-03-20 20:02:29,224 - octoprint.plugins.SlicerEstimator - DEBUG - Metadata line ignored because of it's length.
2024-03-20 20:02:29,227 - octoprint.plugins.SlicerEstimator - DEBUG - Metadata line ignored because of it's length.
2024-03-20 20:02:30,552 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Benchy_1.gcode
2024-03-20 20:02:30,556 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /usr/local/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /octoprint/octoprint/uploads/Benchy_1.gcode
2024-03-20 20:02:30,567 - octoprint.plugins.SlicerEstimator - DEBUG - File: Benchy_1.gcode - metadata written: {'allow_empty_layers': '0', 'arc_fitting': '1', 'arc_fitting_tolerance': '5%', 'avoid_crossing_not_first_layer': '1', 'avoid_crossing_perimeters': '0', 'avoid_crossing_perimeters_max_detour': '0', 'avoid_crossing_top': '1', 'bed_custom_model': '', 'bed_custom_texture': 'Downloads/Anycubic-Bed-Texture.png', 'bed_shape': '0x0,210x0,210x210,0x210', 'bed_temperature': '45', 'before_layer_gcode': ';BEFORE_LAYER_CHANGE\\nG92 E0.0\\n;[layer_z]', 'between_objects_gcode': '', 'bottom_fill_pattern': 'monotonic', 'bottom_solid_layers': '5', 'bottom_solid_min_thickness': '0', 'bridge_acceleration': '1000', 'bridge_angle': '0', 'bridge_fan_speed': '80', 'bridge_fill_pattern': 'rectilinear', 'bridge_flow_ratio': '100%', 'bridge_internal_acceleration': '0', 'bridge_internal_fan_speed': '-1', 'bridge_overlap': '90%', 'bridge_overlap_min': '80%', 'bridge_speed': '25', 'bridge_speed_internal': '150%', 'bridge_type': 'nozzle', 'bridged_infill_margin': '200%', 'brim_acceleration': '0', 'brim_ears': '0', 'brim_ears_detection_length': '1', 'brim_ears_max_angle': '125', 'brim_ears_pattern': 'concentric', 'brim_inside_holes': '0', 'brim_per_object': '0', 'brim_separation': '0', 'brim_speed': '50%', 'brim_width': '0', 'brim_width_interior': '0', 'chamber_temperature': '0', 'clip_multipart_objects': '1', 'color_change_gcode': 'M600', 'complete_objects': '0', 'complete_objects_one_skirt': '0', 'complete_objects_sort': 'object', 'cooling_tube_length': '5', 'cooling_tube_retraction': '91.5', 'curve_smoothing_angle_concave': '0', 'curve_smoothing_angle_convex': '0', 'curve_smoothing_cutoff_dist': '2', 'curve_smoothing_precision': '0', 'default_acceleration': '1000', 'default_fan_speed': '65', 'default_filament_profile': '"Generic PLA @MEGA"', 'default_print_profile': '0.15mm QUALITY @MEGA', 'default_speed': '100', 'deretract_speed': '40', 'disable_fan_first_layers': '1', 'dont_support_bridges': '1', 'draft_shield': 'disabled', 'duplicate_distance': '6', 'end_filament_gcode': '"; Filament-specific end gcode"', 'enforce_full_fill_volume': '1', 'enforce_retract_first_layer': '1', 'ensure_vertical_shell_thickness': '0', 'exact_last_layer_height': '0', 'external_infill_margin': '150%', 'external_perimeter_acceleration': '0', 'external_perimeter_cut_corners': '0%', 'external_perimeter_extrusion_change_odd_layers': '0', 'external_perimeter_extrusion_spacing': '', 'external_perimeter_extrusion_width': '0.45', 'external_perimeter_fan_speed': '-1', 'external_perimeter_overlap': '100%', 'external_perimeter_speed': '40', 'external_perimeters_first': '0', 'external_perimeters_hole': '1', 'external_perimeters_nothole': '1', 'external_perimeters_vase': '0', 'extra_loading_move': '-2', 'extra_perimeters': '1', 'extra_perimeters_odd_layers': '0', 'extra_perimeters_overhangs': '0', 'extruder_clearance_height': '35', 'extruder_clearance_radius': '60', 'extruder_colour': '#808080', 'extruder_fan_offset': '0%', 'extruder_offset': '0x0', 'extruder_temperature_offset': '0', 'extrusion_axis': 'E', 'extrusion_multiplier': '0.87', 'extrusion_spacing': '', 'extrusion_width': '0.45', 'fan_below_layer_time': '100', 'fan_kickstart': '0', 'fan_percentage': '0', 'fan_printer_min_speed': '0', 'fan_speedup_overhangs': '1', 'fan_speedup_time': '0', 'feature_gcode': '', 'filament_colour': '#3584E4', 'filament_cooling_final_speed': '3.4', 'filament_cooling_initial_speed': '2.2', 'filament_cooling_moves': '4', 'filament_cooling_zone_pause': '0', 'filament_cost': '18.5', 'filament_custom_variables': '""', 'filament_density': '1.24', 'filament_diameter': '1.75', 'filament_dip_extraction_speed': '70', 'filament_dip_insertion_speed': '33', 'filament_enable_toolchange_part_fan': '0', 'filament_enable_toolchange_temp': '0', 'filament_load_time': '0', 'filament_loading_speed': '28', 'filament_loading_speed_start': '3', 'filament_max_overlap': '100%', 'filament_max_speed': '0', 'filament_max_volumetric_speed': '10', 'filament_max_wipe_tower_speed': '0', 'filament_melt_zone_pause': '0', 'filament_minimal_purge_on_wipe_tower': '15', 'filament_retract_length': '4', 'filame [... cut ...] im_width': '2', 'wipe_tower_no_sparse_layers': '0', 'wipe_tower_rotation_angle': '0', 'wipe_tower_speed': '80', 'wipe_tower_width': '60', 'wipe_tower_wipe_starting_speed': '33%', 'wipe_tower_x': '180', 'wipe_tower_y': '140', 'wiping_volumes_extruders': '70,70', 'wiping_volumes_matrix': '0', 'xy_inner_size_compensation': '-0.05', 'xy_size_compensation': '-0.05', 'z_offset': '0', 'z_step': '0.005'}
2024-03-20 20:02:30,587 - octoprint.plugins.SlicerEstimator - DEBUG - [['T0', None, 548, 17839]]
```

NilsRo, 2024-03-21:
Interesting...logfile said Slicer is detected successfully. Do you see the metadata in OctoPrint? Can you upload the octoprint log also. You can upload it as a file by drag and drop. Perhaps it a problem with JavaScript in the browser to exclude this I perhaps have to add a new debugging output but let us analyse it step by step.

CraftoHohenvels, 2024-03-21:
This is part of the log. It's the only relevant part. The metadata is not visible.

NilsRo, 2024-03-25:
I added some debug output but I still cannot reproduce it. Could you enable the development branch in the plugins settings, then you have some enhanced logging for the event bus. Octoprints analysis is starting later than yours perhaps it blocks something...but at first I like to know some content of variables that are printed our now.

NilsRo, 2024-03-26:
Thanks a lot. I added a debug message "Plugin message:" that is fired if a dialogbox is triggered like in thhe screenshot in the first post. I cannot see that debug message, could you check if it is logged at another step. Is it shown during this test also?

Is something shown here? Then the metadata is correctly written...
[image]

If you do not find any line with "Plugin message: no_slicer_detected" then the dialog is not triggered by the python backend. Could you check in the development tools of the browser if an error is shown. In Edge it looks like this:
[image]

closed 2024-04-29 by NilsRo