NilsRo/OctoPrint-SlicerEstimator #75: 1.6.5 error on file import in octoprint

fonso2616, 2024-05-02:
## Description
updated plugin to 1.6.5

1. in octoprint, files section, download any existing gcode to your computer
2. try to import the file that was just exported and you get error

i reverted back to 1.6.4 and its working ok now, so something broke in 1.6.5

Describe your problem? error on file import
What is the problem? updated octoprint plugin to 1.6.5 and can no longer import files
Can you recreate it? yes
Did you try disabling plugins? yes and import successfull when disabled
What slicer are you using? cura
Did you uploaded the GCODE file causing the issue? no, any gcode file can be used to re-create
Did you remember to update the subject?**

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Firmware Updater[firmwareupdater] v1.14.0
- M73 Progress Plugin[m73progress] v0.2.1
- OctoText[OctoText] v0.3.3.1
- Slicer Estimator[SlicerEstimator] v1.6.4

**Settings**
- average_prio: true
- metadata_filelist: true
- metadata_filelist_align: top
- metadata_list: 
- metadata_printer: true
- metadata_slicer: true
Installed plugins: (SlicerEstimator)
- useDevChannel: false
- use_assets: true

**Software versions**
- OctoPrint 1.10.0
- Python 3.9.2
- OctoPi* 1.0.0 (build 2023.10.09.151442)

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/124.0.0.0 Safari/537.36

NilsRo, 2024-05-02:
Hi.

I tried the same and cannot create any error uploading a "changed" file again. Could you create a log (set debugging) and perhaps upload the file causing the error you see.

As I understood it happens to previously downloaded files only?

[image]

fonso2616, 2024-05-02:
[image]

[octoprint.log](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/15188860/octoprint.log)

added .log to gcode file for upload
[ball.gcode.log](https://github.com/NilsRo/OctoPrint-SlicerEstimator/files/15188869/ball.gcode.log)

NilsRo, 2024-05-02:
Found the reason and fixed it in 1.6.6. Thanks for the logs and the gcode file.

fonso2616, 2024-05-02:
Great, that was quick.  Ill update later and test it out