NilsRo/OctoPrint-SlicerEstimator #78: Feature Request: Add support for Snapmaker Luban

firebadnofire, 2024-08-26:
## Feature Request: Add support for Snapmaker Luban

I am requesting support for Snapmaker's [Luban](https://github.com/Snapmaker/Luban) slicer software as it is what I use and it currently lacks support

NilsRo, 2024-08-26:
Hi. Could you upload an example GCode. Please activate remaining time calculation if possible, as it is sometimes necessary for some slicers (e.g. prusa).

firebadnofire, 2024-08-31:
Here's an example gcode from Luban
[example.zip](https://github.com/user-attachments/files/16822975/example.zip)

NilsRo, 2024-09-01:
Please activate the development branch in the settings tab of the plugin and run some tests. I added support there for Luban with time estimation and comment (metadata) extraction.

firebadnofire, 2024-09-02:
It looks like it works 👍

NilsRo, 2024-09-05:
Luban support is released...

closed 2024-09-05 by NilsRo