NilsRo/OctoPrint-SlicerEstimator #82: Time to filament change not displayed

FlyingRatBull, 2025-01-25:
I'm using Slicer Estimator 1.6.7 on OctoPrint 1.10.3 and when uploading/printing a file (created with Cura 5.9) with an M600 command, the time to layer change is not shown:

[image]

[i3p-08mm-03mm-600mms_Fillet Radius Finder (Outer) - Design Reference Ruler.zip](https://github.com/user-attachments/files/18546477/i3p-08mm-03mm-600mms_Fillet.Radius.Finder.Outer.-.Design.Reference.Ruler.zip)

[octoprint.log](https://github.com/user-attachments/files/18546521/octoprint.13.log)

NilsRo, 2025-01-27:
I found the issue. It is related that the line is delivered by OctoPrint for parsing in the plugin is not in the expected format. It happens with all files also my old test gcodes. So I like to ask it in the Developer Forum if the interface definition changed somehow before I change the handling. Perhaps it is a kind of bug that have to be fixed in OctoPrint itself.

NilsRo, 2025-01-27:
You can switch do the Development Branch in the settings to test the change. I fixed the issue, was related to my code.

NilsRo, 2025-01-29:
I released the fix as it is a general issue. You can simply run an forced update to get the newest version if you like.

closed 2025-01-29 by NilsRo