NilsRo/OctoPrint-SlicerEstimator #84: Add Time to Pause (M601)

FlyingRatBull, 2025-02-20:
It would be nice to have a "Time to Pause" in addition to the "Time to Filament change" as this also requires user input at that time and thus would be nice to know, when to expect it.

In _PrusaSlicer_ it is done via an `M601` command, however the _Marlin_ documentation also lists `M0`/`M1` for an analogous purpose.

Either way I think it would also help if the text given to the command (e.g. "Insert bearings") would be shown either directly or in a tooltip. However this is done in two different ways, as it seems:
* _PrusaSlicer_ puts an `M117 [MESSAGE]` before the `M601`
* _Marlin_ documentation shows `M0 [MESSAGE]` to be valid

I know that `M601` is used by various printers/slicers and `M0` seems to be used by at least _[Cura (plugins)](https://all3dp.com/2/cura-pause-at-height-how-to-do-it/)_.

[Example GCode with single M601 after M117.zip](https://github.com/user-attachments/files/18883826/0.9mm.-.yourDroid.PLA.-.Mainboard.Fan.Cover.90mm.zip)

NilsRo, 2025-02-20:
It make sense to add this. Good idea. I will take a look...but I am not sure to support M117 as there are plugins available to show this as a toolbox also in this case.

FlyingRatBull, 2025-02-21:
Regarding a notification with the M117 message I agree, other plugins do this already. However I propose to integrate it more of in a "timeline" way:

[image]

NilsRo, 2025-02-21:
Please switch to development branch in the settings and try if it works like expected. I added M0 and M601 to the list actually without parsing comments.

FlyingRatBull, 2025-02-22:
Seems to work just fine, also tested with multiple M601 commands per print.

Is there a functionality to display the estimated time (on the clock) for each of the steps? Like for example in a tooltip. Because I guess 0.5 hours could be something between 15 and 45 minutes depending on the used rounding.

NilsRo, 2025-02-22:
I added it like it is done in the OctoPrint standard as a tooltip. You must over over it with the mouse for some seconds...

FlyingRatBull, 2025-02-23:
I tried that (with a virtual printer) but neither on the "pause (M601)" nor on the ETA was there a tooltip available.

FlyingRatBull, 2025-02-23:
With the current update it shows the time when not printing (although it just seems to be the upload time + ETA or something else?) - It was 7pm and a step in "2.5 hours" had a tooltip with "02:22".

However once I started printing (on the virtual printer) and "Print Time" had a value, no tooltip was present.

NilsRo, 2025-02-24:
That are 2 hours and 22 minutes left. It is the same behavior like with the remaining print created by OctoPrint. 

I cannot reproduce it, the tooltip is available every time here. Are you perhaps using some other plugins changing the behavior if the tooltips? I am reusing some code provided by OctoPrint. Could you check if there are some JavaScript error in the browser console (F12).

[image]

FlyingRatBull, 2025-02-24:
My fault, I thought it would be the time of day.

I tested it again and the behavior is weird... when printing, the tooltip doesn't work although the HTML stays exactly the same and there are no errors in the console. I'd guess it has something to do with the virtual printer or another plugin.

Thus I'd say it's on my end and your code is fine 👍🏻

NilsRo, 2025-02-24:
You can disable all plugins without uninstalling for testing if you like. Perhaps we can identify the reason. Actually I am testing with virtual printer only as my Mega S is out of order.

FlyingRatBull, 2025-02-24:
I did that with the same result:

* With the model loaded: Tooltip visible
* With print started (not yet at temperature): Tooltip visible
* With tool at temperature (Layer progress reset to0%): Tooltip **not** visible
* With print paused: Tooltip visible

It seems to be a problem with Firefox (I'm using 135.0.1 (64-bit) on Ubuntu 22.04) as with Chromium it works fine.

NilsRo, 2025-02-24:
Is the OctoPrint tooltip still visible - remaining print time? If yes it's perhaps some slight variance in my code. If not it's related to browser behaviour or OctoPrint coding.

FlyingRatBull, 2025-02-24:
The "Print Time Left" tooltip works just fine in both Browsers so it seems to be something in your code.

NilsRo, 2025-02-25:
The update behavior in JavaScript Knockout was not optimal and triggered too many refreshes...leads into higher resource consumption and also some browsers does not optimize it like Chrome Engine does. I changed it now to a lightweight event based update...it is not finished but could you check if everything behave fine now? My tests with Firefox are fine now...

FlyingRatBull, 2025-03-05:
Sorry for the late answer, I had some other troubles with my Octopi preventing me from testing this.

The tooltip now works as expected 😄

NilsRo, 2025-03-05:
Perfect. Still spend some thoughts how the 'Message' part could work but will not have the time to implement it soon. But we can leave this request open to have an eye on it.

FlyingRatBull, 2025-03-05:
That would be nice!