NilsRo/OctoPrint-SlicerEstimator #86: Error from plugin when trying to estimate printing time. - Creality Slicer

dorminyk, 2025-05-10:
## Description

**ENTER DESCRIPTION HERE

Describe your problem? 
          "Slicer not detected. Please open a ticket if the slicer should be known..."

What is the problem? 
          When upload a Gcode file produced from the Creality slicer, It shows me the above message error even though I've got "Verbose Gcode enabled in the slicing Software

Can you recreate it?
          Every time

Did you try disabling plugins?
          Yes

What slicer are you using?
          Creality Slicer.

Did you uploaded the GCODE file causing the issue?

[Ironing Stack - Test Print - Key Chain_ABS_25m17s.gcode.zip](https://github.com/user-attachments/files/20140019/Ironing.Stack.-.Test.Print.-.Key.Chain_ABS_25m17s.gcode.zip)

Did you remember to update the subject?**

**Plugins installed**
- Cancel Objects[cancelobject] v0.6.0
- Cost Estimation[costestimation] v3.6.0
- Crealitycloud Plugin[crealitycloud] v1.0.9
- Cura Thumbnails[UltimakerFormatPackage] v1.0.2
- Dashboard[dashboard] v1.19.12
- Detailed Progress[detailedprogress] v0.2.7
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Exclude Region[excluderegion] v0.3.2
- Fan Speed Control[fanspeedslider] v0.2.3
- Fan Speed Mirror[FanSpeedMirror] v1.3.1
- FilamentBuddy Plugin[filamentbuddy] v0.1.0
- FileManager[filemanager] v0.1.6
- Firmware Updater[firmwareupdater] v1.14.1
- Fullscreen Plugin[fullscreen] v0.0.6
- Marlin EEPROM Editor[eeprom_marlin] v3.3.0
- OctoApp[octoapp] v2.1.13
- OctoEverywhere![octoeverywhere] v4.1.3
- OctoPod Plugin[octopod] v0.3.18
- Octoprint Display ETA[display_eta] v2.2.0
- Preheat Button[preheat] v0.8.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.4.0
- Resource Monitor[resource_monitor] v0.3.16
- Simple Filament Change Buttons[simple_filament_change_buttons] v1.2
- Slicer Estimator[SlicerEstimator] v1.6.9
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.8
- SlicerSettingsParser[SlicerSettingsParser] v3.3.0
- SmartABL[SmartABL] v1.2.2
- SpoolManager Plugin[SpoolManager] v1.7.7
- STL Viewer[stlviewer] v0.4.2
- Tab Order[taborder] v0.5.12
- Tempsgraph Plugin[tempsgraph] v0.3.7
- Themeify[themeify] v1.2.2
- WebcamExtras Plugin[webcamextras] v0.0.12

**Software versions**
- OctoPrint 1.11.0
- Python 3.11.2

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/18.4 Safari/605.1.15

chadberg, 2025-06-12:
Sam using CrealityPrint

NilsRo, 2025-06-12:
Please switch to development version (in the settings of the plugin) and try if it is working now with Creality Print.

chadberg, 2025-06-12:
Not yet. When I get home and the current print completes I will test and let you know.

chadberg, 2025-06-14:
Yes, it appears to be working just fine now. Checkmark and all.

closed 2025-06-24 by NilsRo